MacRover/maxwell #11: RAD Board motor operation

emcauliffe, 2024-01-27:
Validate that RAD can control stepper motor

emcauliffe, 2024-02-04:
Issue: no continuity between A+ and BMA 1 or Q1 drain

emcauliffe, 2024-02-04:
> Issue: no continuity between A+ and BMA 1 or Q1/Q4 drain

Resolved by reflowing solder on header

ishanvermani, 2024-02-04:
Issue: Motor does not respond in standalone mode with very basic software. Have probed continuity and relevant pins to attempt to determine issue with no success. Motor is just 'freewheeling' which suggests no current passing through coils as well as minimal current draw from power supply.

emcauliffe, 2024-02-06:
> Issue: Motor does not respond in standalone mode with very basic software. Have probed continuity and relevant pins to attempt to determine issue with no success. Motor is just 'freewheeling' which suggests no current passing through coils as well as minimal current draw from power supply.

Manually jumped PA15 to 3V with careful placement of a jumper wire. Motor makes noises but does not seem to step.

For future revisions, recommendation is swap PA15 with PA8.

emcauliffe, 2024-02-12:
Confirmed RAD motor control by setting registers over SPI and using step/dir for operation. Torque is low, but this should be resolved by figuring out what the appropriate motor control settings are.

emcauliffe, 2024-02-13:
Reopening because motor spins but only when the step size register `MRES` is set to an undefined state of `1111`

emcauliffe, 2024-02-15:
Issue was that the LSB (least significant byte) was being sent to the stepper driver first. Should be MSB first. Fixed and now the motor works great

closed 2024-02-15 by emcauliffe