MacRover/maxwell #56: EEPROM storage and retrival of Device Data

ishanvermani, 2024-09-06:


sullynumber9, 2024-09-17:
- Successfully able to break down a struct into a byte array, and reconstruct and read back those same values
- Moving on to sending through EEPROM.

sullynumber9, 2024-09-17:
- Writing to EEPROM successful.  Working on reading back from EEPROM

sullynumber9, 2024-09-17:
- Able to send data, but are not sure if it is being written to and/or read back properly.  Issue currently with read/write

sullynumber9, 2024-09-17:
- EEPROM is organized 32 pages of 16 byte chunks.  It is not possible to send more than 16 bytes to EEPROM at once
- Code must be written and read to/from EEPROM in 15 byte chunks, plus byte 0 as the EEPROM address
- For next time in Hatch: Creating a function to take the split data in an array and send the data in 16 byte chunks
- Elec coming in at 6:30 to finish boards, we will also come in then on Wednesday

sullynumber9, 2024-09-19:
- Created a function to organize EEPROM into 16 byte chunks for writing data.  Function to be tested at a later date

Next steps:
- Creating the read function (similar, but with different data sizes due to not reading the EEPROM address back
- Creating separate functions for read_raw and write_raw

sullynumber9, 2024-09-19:
- Able to break down the struct, send the byte array into EEPROM, and read the byte array back
- One byte is lost in the process (only able to write 15 bytes per page)
- Currently troubleshooting getting the byte array back into the struct (math is not correct for the last iteration)

vaibhav-gopal, 2024-09-19:
- Fixed losing the byte in progress, still only able to write 15 bytes per page
- Fixed getting byte array back into a struct, replaced for loop in write with memcpy for performance, and fixed the memcpy implementation in read
- Changed functions to take page address instead of byte address to prevent data loss
- Tested with different addresses, read and write, with multiple values, struct is returned whole with _no_ data loss (floats have some rounding errors after the 6th decimal)

closed 2024-09-22 by ishanvermani