MacRover/maxwell #59: Limit Switch integration (mechanical LS and Software Bound)

ishanvermani, 2024-09-06:


sullynumber9, 2024-09-24:
- Wrote preliminary code to determine actions based on the limit switch being triggered
- Code written to block movement out of bounds
- Rollovers tracked according to project requirements
- Code currently untested, testing code to begin as soon as possible
- Code is located in feature/RAD-LS > Firmware > RAD > Release > main.c within the while loop

sullynumber9, 2024-09-25:
- Updated code to fix projected bugs with the encoder value and last encoder value variables.  No testing has been done at this time

sullynumber9, 2024-09-27:
- Added a small number of bug fixes, no testing done yet (May have to increase task end time)

sullynumber9, 2024-10-03:
- CCW and CW direction were swapped compared the original code (Need to validate this on the rover to see if it is causing issues)
- Encoder sometimes resets back to 0 unexpectedly (not sure if this is an issue with how well the encoder is attached)
- Altered lines 290 and 310 to include a check for minimum rotations as well as activation of the limit switch

RAD_Type 1 Testing

No_CW blocker working properly
Rollovers incrementing properly when moving from 330-360 degrees to 0-30 degrees (+1) near both the maximum and minimum limits
Rollovers decrementing properly when moving from 0-30 degrees to 330-360 degrees (-1) near both the minimum and maximum limits
No_CCW blocker working properly
Hitting the limit switch properly sets rollovers to 20 and CCW blocker to 1 (activates the blocker)

TODO: Test RAD_Type 2

vaibhav-gopal, 2024-10-08:
- Updated code to use Ethan's PID library rollover counter implementation and PID_SetZeroPoint

- Added a calibration step to ensure stepper can only move towards limit switch to calibrate first

- Main functionality now consists of checking if we've hit the limit switch (GPIO Pin), and checking if our current angle has exceeded the software stop that was calculated during the initial calibration step

closed 2024-10-27 by sullynumber9