MacRover/maxwell #70: Existing RAD Validation

ishanvermani, 2024-09-09:
RAD 7:
- Working CAN

sullynumber9, 2024-09-09:
Stepper control for RAD 7 working (Tested without encoders)

sullynumber9, 2024-09-10:
RAD 4:
- Can communicate using STLink
- CAN working
- Basic motor control working (no encoders)
- EEPROM working (can read and write different hex values to EEPROM - much quicker performance than RAD 4)
Note: J5 works, but appears slightly loose

RAD 7:
- EEPROM not working (identical code to that used on RAD 4)

sullynumber9, 2024-09-11:
RAD 7: Issue observed with the pull up resistor connection - shortcircuit between two of the pins (potentially why EEPROM isn't working)

sullynumber9, 2024-09-11:
RAD 7: Shortcircuit between VCC3V3 and SDA fixed - EEPROM still not working

Next steps: Asking elec to replace the EEPROM chip.  Could also be an issue with the STM32 chip

sullynumber9, 2024-09-11:
RAD 4: Encoder working, can read angle values from encoder.  90 degree turn tested accurately

sullynumber9, 2024-09-11:
RAD 7: Encoder working, can read angles from encoder.  90 degree turn tested accurately.

sullynumber9, 2024-09-11:
RAD 7: 
- PID responding to changes
- Limit switch tested and working - both limit switch pins working

RAD 4:
- PID responding to changes
- Limit switch tested and working - both limit switch pins working
- Reflow may be required on J7...issues connecting to the board, but works when the wires from the STLink

sullynumber9, 2024-09-18:
RAD 4:
- Reflow on the board, can consistently communicate

RAD 7: 
- EEPROM chip replaced, now working

closed 2024-09-18 by sullynumber9