MacRover/maxwell #72: RAD Validation of repaired boards

chosterto, 2024-09-27:


chosterto, 2024-09-27:
Can successfully connect to the STM32 MCU for RAD 6

nickoenig37, 2024-09-30:
CAN works on RAD 6

chosterto, 2024-09-30:
Basic STEP/DIR motor control for RAD 6 working

sullynumber9, 2024-10-01:
RAD 6: Successfully able to read and write values to EEPROM

sullynumber9, 2024-10-01:
RAD 6: PID & Encoder Working

sullynumber9, 2024-10-01:
RAD 6: Both limit switches working

sullynumber9, 2024-10-07:
RAD 2:
- PID/Basic Stepper control not working
       - Motor was vibrating, but was not able to properly rotate in the same way as the other motors
- Issues with basic connection (board seemed to be connecting and disconnecting seemingly at random)
- Limit switches not working (Continuity tested between ground and the limit switches as working)
- EEPROM send/receive working
- CAN not working

RAD 4:
- Not powering on (I tried to power this on to test if it was a code issue with RAD7, and the board would not power on.  The green status LED blinked twice as I shut the power supply off but the board would not power on.  The power supply also wouldn't power on afterwards until I adjusted the plug at the power bar.  Not sure what the issue was as the board did seem to be connected properly.)

sullynumber9, 2024-10-08:
RAD4:
- Need reflow on J5

RAD2:
- Tested CAN again, still not working (CAN also not working on RAD7, so appears as a software issue)
- Basic step/dir control not working (Motor will not complete a full rotation, and will only vibrate.  This exact code was tested on RAD7 as well)

CAN Testing (on RAD 7):
- Using validation code and code in the main branch (from comp) --> no change
- Testing for continuity between CAN-L from the PCAN and CAN-L from the transceiver (successful)
- Repeated the above for CAN-H (successful)
- Tested blinking an LED from within the Broadcast_RAD_Message function (successful)
- Tested CAN with the EEPROM resistor on the board (unsuccessful)

ishanvermani, 2024-11-05:
RAD 4, 6, 7, and 10 are all operational

closed 2024-11-05 by ishanvermani